ms-iot/content #1390: Dragonboard 410c analog audio output

endyb66, 2018-08-08:
On the Dragonboard 410c, there is a analog audio header (J7). When connecting small speakers to the direct output SPKR_L and SPKR_R, everything works. But i need to attach a 3.5mm jack, sadly on These pins is no Audio Signal and there is no audio device listed to Change to.

saraclay, 2018-08-10:
Hi @endyb66 - what exactly is your question here? What information are you looking for?

saraclay, 2018-08-20:
Hi @endyb66, have you looked into the hardware modifications recommended here? https://developer.qualcomm.com/qfile/29468/lm80-p0436-43_c_stereo_connector_and_audio_routing_on_dragonboard_410c_application_note.pdf

endyb66, 2018-08-20:
@saraclay, yes i have, but it simply does not work. I read in the document there, that for e.g. on linux based systems it has to be modified something,

endyb66, 2018-08-21:
@saraclay, i have only one output device listed there.

saraclay, 2018-08-21:
Could you file a bug on our forum? Apologies - https://social.msdn.microsoft.com/Forums/en-US/fad1c6a0-e578-44a7-8e8d-95cc28c06ccd/need-logs-if-your-device-hasnt-updated-to-the-latest-iotcore-version?forum=WindowsIoT

closed 2018-08-21 by saraclay

endyb66, 2018-08-22:
@saraclay, couldn't you reproduce the issue?